cucapra/dahlia #223: Code generation rounding off values

sa2257, 2019-07-24:
Generated code for `let PI: double = 3.1415926535` rounds off to `double PI = 3.1415927`.

rachitnigam, 2019-07-24:
Changing the EFloat syntax node to take in Double instead of Float should fix it.

sa2257, 2019-07-24:
Hmm.. changed https://github.com/cucapra/seashell/blob/254f3b543e7ffe17d88620626529b2f9aed4f632/src/main/scala/common/Syntax.scala#L108 `float` to `double`. It gave me `double PI = 3.1415927410125732` :confused:.

rachitnigam, 2019-07-24:
Jesus Christ. What is this mess. I’ll take a look.

closed 2019-07-24 by rachitnigam

sampsyo, 2019-07-25:
That's a tricky solution, but it'll do! Nice work.

rachitnigam, 2019-07-25:
The issue with just changing the AST node to hold double was that parser was still using toFloat.

Changing that to toDouble also makes the whole thing work but we don’t inspect floating values right now so no reason to keep them in the AST.

commit 45caf876 (2019-07-24): floating point numbers are represented as strings in the AST. Fixes #223.