cucapra/dahlia #230: Partitioning arguments insert pragmas twice, once inside function arguments.

sa2257, 2019-07-31:
Check [this](http://gorgonzola.cs.cornell.edu:8000/jobs/fD9nZ-W5U-w/files/code/gemm.cpp) where pragmas are placed in the function arguments which shouldn't happen.

sampsyo, 2019-07-31:
That looks bad! Here's the code you're referring to:

```
#pragma SDS data copy(m1[0:64][0:64])
void gemm(double m1[64][64]

#pragma HLS ARRAY_PARTITION variable=m1 cyclic factor=8 dim=2, double m2[64][64]

#pragma HLS ARRAY_PARTITION variable=m2 cyclic factor=8 dim=1, double prod[64][64]
) {
  
  #pragma HLS ARRAY_PARTITION variable=m1 cyclic factor=8 dim=2
  
  #pragma HLS ARRAY_PARTITION variable=m2 cyclic factor=8 dim=1
```

These should *just* go in the body, right?

At the risk of stating the obvious, the problem here is that the `double m2[64][64]` declaration is on the same line as the `#pragma` for `m1`, so the compiler can't see it.

rachitnigam, 2019-07-31:
Bug in the code gen phase (damn, that module is not well written :-( ).

Issue across two files: `CppLike` uses the `emitDecl` function to generate C++ types for the arguments. `emitDecl` uses `emitArrayDecl` which is overridden in `VivadoBackend` to generate a banking pragma after an array (this is used to support local arrays) which means bank pragmas are generated for arguments too.

The core issue is that pragmas are generated differently for function argument types than they are for local variables but the code generation treats them the same (by overriding the meaning of `Decl`s in multiple contexts).

Feel free to take a hack at it if you want. If not, I'll fix it tonight. One good use case for @ViviYe's new backed is running a simple code validation pass on the run-tests every night to make sure the compiler isn't broken.

Longer term, the RTL backend will need testing through synthesis flows so this is somewhat worth thinking about.

commit 4d13a9a7 (2019-08-05): Fix #230: put partitioning in the right place

Now partitioning pragmas go in the function header and `let` rather than
being part of the declaration This solves both #230 and a related
problem where pragmas following `let` declarations would go before the
initializer or semicolon.
closed 2019-08-05 by sampsyo