cucapra/dahlia #238: Possible read capabilites bug

rachitnigam, 2019-08-15:
Should this program type check?

```
decl A: float[10][10 bank 5];

for (let i = 0 .. 10) unroll 5 {
  for (let j = 0 .. 10) {
    A[j][i];
  }
}
```

The read can certainly be implemented so that it's shared across all the internal for loops.

sampsyo, 2019-08-17:
Seems like yeah? Inasmuch as this not-quite-equivalent program should be allowed:

```
for (let i = 0 .. 10) unroll 5 {
  A[0][i]
  ---
  A[1][i]
  ---
  ...
}
```

(It's not quite equivalent because this version uses more resources than the original, but the timing is the same.) But perhaps I missed the point here…

rachitnigam, 2019-08-22:
Looking harder at this post-deadline, it's not hard to see why it works fine.

closed 2019-08-22 by rachitnigam