cucapra/dahlia #255: Adding literal to represent fixed type

tissue3, 2019-09-23:
Fixed type is tricky since 0.2 has no way to be represented as precise binary number.
So we will add a new inference rule for the syntax "Let". Using "f'0.5" means the number is going to be represented in minimal number of required, <1,0> in this case. If a number is not able to be represented by fixed point, the program will give an error.
On the other hand, if one does not specify the literal"f'", the implicit inference will take the number as double if there is a decimal point. For example, "Let x = 0.5" says x will be inferred as 0.5 by default.

rachitnigam, 2019-09-24:
I'm still in favor of having separate fixed point literals. More precisely (ha):
- `let x = 1.2` gives `x` the type `double`.
- `let x = f1.25` gives the smallest fixed point type that can represent it. However, fixed point representation might be unintuitive, so we could also take the more radical approach of exactly specifying the size of the literal using `f<1,2>1.2` since we generate slightly better error messages for this. This also reminds me of how P4 implements [sized integers](https://p4.org/p4-spec/docs/P4-16-v1.1.0-draft.html#sec-integers) (section 7.1.6.6). 
- `let x = f1.2` errors out since there is no fixed for representation for it.

One thing to realize here is that no matter what solution we implement (removing type inference vs. sized literals), fixed points will be cumbersome to use because a Dahlia program will be tied to it's choice of double vs. fixed point at some level.

The ideal solution would require a thoughtful redesign of the type system to have bitwidth polymorphism (which is surprisingly hard, as the P4 team learned). This is just a stepping stone for that.

sampsyo, 2019-09-24:
Here's a funky idea that might be terrible. The idea would be to let both of these assignments, which both involve rounding, to work the same way:

```
let x: double = 1.2;
let y: fix<1,2> = 1.2;
```

Literals could have a new type for representing exact numbers. You could call it `Rational`, for example, because it can precisely represent any number with a finite number of decimal (or other base?) digits. Rationals can be freely converted to any approximate number type, including `double`, `fix<N, M>`, perhaps even `bit<M>`, and perhaps a future generalized `float<N, M>`.

Rationals must _always_ be compile-time constants, exactly like our current `TStaticInt`, because they can't in general be represented at run time. Instead, when they get converted to some other number type, we can statically do the conversion and emit the concrete value of the appropriate type.

If you want to use a literal as part of a larger expression, you use casting to describe the type you want:

```
let x = (1.2 as double) + some_function();
let y = (1.2 as fix<1,2>) + other_function();
```

Relying on casts obviates the need for us to cook up special literal syntax for each numeric type. For example, this would free us from coming up with a prefix to distinguish `float` and `double` eventually, and perhaps even arbitrary-precision floating-point numbers.

For bonus points, we could even support compile-time arithmetic on these constants.

rachitnigam, 2019-09-27:
Fixed by #257.

closed 2019-09-27 by rachitnigam