cucapra/dahlia #263: Conflict in error message

tissue3, 2019-10-18:
The following codes looks the same to me:
```
          decl a: bit<32>[10 bank 5];
          for (let i = 0..10) unroll 5 {
            if (a[i] > 10) {
              a[i] := 1
            }
          }
```
```
          decl a: bit<32>[10 bank 5];
          for (let i = 0..10) unroll 5 {
            let x = a[i]
            ---
            if (x > 10) {
              x := 1
            }
          }
```
The first one cannot pass the capability checker, throwing InvalidCap error. The latter passes capability checker and throws loop dependency error.

rachitnigam, 2019-10-18:
This is expected behavior.

In the first program, you have a read from `a` in `a[i] > 10` and a write to `a` in `a[i] := 1`. In the second program, you are writing reading an writing to `x` which doesn't point to the memory `a[i]`.

To be clear, HLS will need to insert a `---` b/w the read and the write of `x` to preserve semantics.

rachitnigam, 2019-10-18:
The `multi-ported` branch gives a better message in this case:
```
[Type error] [4.17] Bank 0 in dimension 1 of physical resource `a' already consumed.
              a[i] := 1
                ^

[3.19] Last consume happened here:
            if (a[i] > 10) {
                  ^
```

tissue3, 2019-10-18:
I see your point. Close the issue.

closed 2019-10-18 by tissue3