cucapra/dahlia #299: [Performance] Rewrite parser

rachitnigam, 2020-06-26:
After a random evening of profiling things trying to see if the Dahlia compiler compiled with [scala-native](https://scala-native.readthedocs.io/en/v0.3.9-docs/) would run any faster (it was slower), it turns out the compiler spends the following % of time in the parser:

- JVM: 45-51%
- Native: 77-96%

Probably worth rewriting the compiler using [Fastparse](https://www.lihaoyi.com/fastparse/) or [Parboiled](https://github.com/sirthias/parboiled) which are reported 100x faster.

Of course, the biggest performance win would be to remove the JVM startup time which might be possible with scala native when we rewrite the parser.

sampsyo, 2020-06-26:
Dang; I'm kinda shocked that scala-native would make it slower! Fastparse claims it's 200x faster than scala-parser-combinators, so uh, that's pretty fast.

rachitnigam, 2020-06-26:
The root cause for the slowdown is the parser is 10x slower with native compared to JVM (!!). So maybe if we remove the parser overhead, we’ll be good.

Scala native also only works for scala 2.11 (were on 2.13) so there might be various regressions.

PR #336 (2020-12-29): Use FastParse for parsing
Fixes #299.
closed 2020-12-29 by rachitnigam