cucapra/dahlia #310: Records + BufferMemoryIO breaks capability checker

sgpthomas, 2020-08-07:
This program doesn't type check with the BufferMemoryIO pass turned on:
```C
record point {
  x: bit<32>;
  y: bit<32>
}

decl shape1: point[2];
decl shape2: point[2];
decl result: point;

let x: bit<32> = 0;
let y: bit<32> = 0;

for (let i: bit<1> = 0..2) {
  let x = shape1[i].x + shape2[i].x;
  let y = shape1[i].y + shape2[i].y;
} combine {
  x += x;
  y += y;
}

let out: point = { x = x; y = y };
result := out;

```

However, this generates the following code, which does type check.
```C
record point {
  x: bit<32>;
  y: bit<32>
}

decl shape1: point[2];
decl shape2: point[2];
decl result: point;

let x: bit<32> = 0;
let y: bit<32> = 0;

for (let i: bit<1> = 0..2) {
  let shape1read0: point = shape1[i];
  let shape2read0: point = shape2[i];
  ---
  let x : bit<32> = (shape1read0.x + shape2read0.x);
  let shape1read1: point = shape1[i];
  let shape2read1: point = shape2[i];
  ---
  let y : bit<32> = (shape1read1.y + shape2read1.y);
} combine {
  x += x;
  y += y;
}

let out: point = { x = x; y = y };
result := out;
```
I don't know where the problem is exactly.

rachitnigam, 2020-08-08:
The input program doesn't type check for a trivial reason: you're re-binding `x` in the for loop. Can you give an example that doesn't type check on master but should?

sgpthomas, 2020-08-08:
I fixed the examples, for some reason I lost all the capitilzation when I copied over the examples. sorry about that. The other thing is that I fixed this by not using separating the reads into a separate logic time step. However, if you change the `construct` function at 'BufferMemoryIO.scala:51' to return `CSeq` instead of `CPar`, then this bug happens. I'm concerned that this pass is changing the AST in some way that breaks future passes. Using `CPar` fixes the issue, but it may not fix the root cause of the issue.

I don't think this is an important issue to solve right now though.

rachitnigam, 2020-08-14:
Doesn't seem to happen on master which includes a part of the BufferMemoryIO pass.

closed 2020-08-14 by rachitnigam