cucapra/dahlia #318: Running `sbt assembly`.

cgyurgyik, 2020-09-23:
When running ```sbt assembly```, I get the following:
```
 % sbt assembly
[info] welcome to sbt 1.3.13 (N/A Java 14.0.1)
[info] loading project definition from /Users/X/project
[info] set current project to X (in build file:/Users/X/)
[error] Expected ';'
[error] Not a valid command: assembly
[error] Not a valid project ID: assembly
[error] Expected ':'
[error] Not a valid key: assembly
[error] assembly
[error]         ^
```

Is there something I'm missing here? I followed everything except the cargo install runt, since I already had runt installed from the futil repo.

rachitnigam, 2020-09-23:
Try: `sbt "; assembly"` or first running `sbt` and then when the sbt terminal start, typing `assembly`.

cgyurgyik, 2020-09-23:
Wrong directory :)

closed 2020-09-23 by cgyurgyik