cucapra/dahlia #323: Changing the name of a FuTIL component

cgyurgyik, 2020-10-30:
Currently, we can use `-n <name>` to change the name of a kernel from `kernel` to `<name>` at the Dahlia level. However, when lowered to FuTIL, the component remains as `main`.

For example: Running this from my Dahlia repository:
``` ./fuse ./../futil/examples/dahlia/vectorized-add.fuse  -n NewName -b futil --lower```
will change the name of the kernel when lowered to Vivado, but in FuTIL it remains unchanged.

sampsyo, 2020-10-31:
Good point! To make the implicit explicit, this is important for doing https://github.com/cucapra/futil/issues/186, which needs to emit & merge non-`main` components.

Here is where the name is hard-coded, FWIW:
https://github.com/cucapra/dahlia/blob/6a92be70fc148e428251446517a96452986e0151/src/main/scala/backends/futil/FutilBackend.scala#L654

and I think it could instead get the name from `config`?

cgyurgyik, 2020-11-01:
I think it would be good to also look into adding arguments and a return to a Dahlia program. This allows me to write a reusable function for Relay2FuTIL lowering.

For example,
I want to call an add on two vectors. Then, I'd like to lower a Dahlia program that takes in two arguments and adds them together. The final group after being lowered to FuTIL would return the summed vector by placing its values in some vector that was also passed in as the "return argument."

Maybe there's a better/cleaner way to do this? Either way, I'd be happy to look into this if someone gives me a push (or slight shove) in the correct direction on where to start.

closed 2020-11-05 by cgyurgyik