cucapra/dahlia #325: Adding input argument(s), output argument to a Dahlia program

cgyurgyik, 2020-11-01:
Similarly for completing https://github.com/cucapra/futil/issues/186:
I think it would be good to look into adding arguments and a return to a Dahlia program. This allows me to write a reusable function for Relay2FuTIL lowering.

For example,
I want to sum two vectors. Then, I'd like to lower a Dahlia program that takes in two arguments and adds them together. The final group after being lowered to FuTIL would return the summed vector by placing its values in some vector that was also passed in as the "return argument."

Dahlia program inputs: `(A: ubit<32>[8], B: ubit<32>[8])`
Dahlia program output: `(C: ubit<32>[8])`
Dahlia program body:
```
for (let i: ubit<4> = 0..8) {
  C[i] := A[i] + B[i];
}
// However, need to actually pass C outside of this component as well. Not sure about this step.
```

Maybe there's a better/cleaner way to do this? Either way, I'd be happy to look into this if someone gives me a push (or slight shove) in the correct direction on where to start.

sampsyo, 2020-11-01:
Good point; this is indeed necessary for integrating Dahlia-emitted components into larger designs. I think the first step is to decide what we actually want the FuTIL to look like to implement these inputs/outputs. Two ideas come to mind:

* Maybe we can reuse FuTIL's "externalized memories," which I think @sgpthomas can explain in more detail. But the rough idea, IIRC, is that they behave like memories but they do not actually exist as subcomponents within the FuTIL component: instead, the way they get implemented is by wiring up the input/output ports of the memories as input/output ports on the containing module. I believe we invented this to synthesize designs with Vivado so the memories could be connected to (imaginary) outside hardware that feeds our design with inputs and collects outputs.
* Maybe we just want to hook these directly up to input/output signals in the component. That is, for every memory like `A` or whatever that forms an interface to the component, we create ports like `A_addr`, `A_read_en`, `A_write_en`, and `A_data`, and these go *on the emitted FuTIL component*. Then the "guts" of the FuTIL component can access an outside memory through these ports. The super-component that uses them is responsible for wiring these ports up to an actual memory it controls. (This is the same spirit as the "externalized memories" in the previous bullet but done explicitly inside the FuTIL universe instead of as a post-facto code-generation detail.)

cgyurgyik, 2020-11-02:
I'm not following your second bullet point. Can you give me an example of what you mean? In either case I'll bring this up in the sync-up tomorrow.

sampsyo, 2020-11-02:
Sure. While a version with an "internal memory" would look like this:

```
component foo() -> () {
  cells {
    A0 = prim std_mem(...);
  }
  ...
  // Eventually, to write into the memory:
  A0.write_en = 1'd1;
  A0.addr = x;
  A0.in = y;
}
```

One with an "external memory" would look like this:

```
component foo() -> (A0_write_en: 1, A0_addr: 6, A0_in: 32) {
  cells {
    // No memory here.
  }
  ...
  // Eventually, to write into the memory, use the ports
  // leading "out of" this component:
  A0_write_en = 1'd1;
  A0_addr = x;
  A0_in = y;
}
```

So basically, take the ports that form the interface to the memory and make them ports on the component itself. (Not sure I got the input/output port syntax right.)

cgyurgyik, 2020-11-02:
If I'm understanding correctly, you're saying for a given input and output, pass them in by "reference," as I've started doing with Relay functions. That's what I was thinking along the lines of as well. 

My assumption is this would need to be done at the Dahlia level, or at least signify that `A0` is an output argument at the Dahlia level. While this addresses an output, it doesn't necessarily address input wires either, unless you mean to apply the same idea to those as well.
__________________________________________________________________________________________________________________________________
My one follow-up question is: How to generalize this for memory of different sizes? This is trivial for a 0d tensor memory.
For example, given a `A0` of size `1 x 1`, we have something like:
```
component foo (x_out: 32, y_out: 32, A0_done: 1) -> (A0_write_data: 32, A0_write_en: 1, A0_addr0: 1) { 
   // Does something with x_out and y_out and places it in A0_write_data.
 }
```

 But, if we want to read all addresses of `A`, and `A` is `1 x 4`, then we'd have to make some changes. However, I'm currently not sure how to encapsulate that at the component signature level. I know we've touched on passing in a `begin` and `end` index, but I'm not sure if/how that would work. This is my high level idea:

```
component foo (x_out: 32, y_out: 32, A0_done: 1, begin:2, end: 2) -> (A0_write_data: 32, A0_write_en: 1, A0_addr0: 2) {
A0_addr0 = begin;
group process_single_index { 
  A0_write_data = x_out + y_out; // :-(
  A0_write_en = 1;
}
group update_idx { A0_addr0++; }
control { while (update_idx < end) process_single_idx; }
}
```
The issue with the above is that this will simply pass in `x_out + y_out` into every value of `A0`. So what we'd rather have is 
```
group process_single_index { 
  A0_write_data = x_at_index_i + y_at_index_i; // How?
  A0_write_en = 1;
}
group update_idx { A0_addr0++; x++; y++; }
```
Is that even possible?

sampsyo, 2020-11-02:
> My assumption is this would need to be done at the Dahlia level, or at least signify that `A0` is an output argument at the Dahlia level. While this addresses an output, it doesn't necessarily address input wires either, unless you mean to apply the same idea to those as well.

Yes and yes: it would need to be done in the Dahlia compiler, and it would need to do an analogous thing to input memories.

For dealing with memories of multiple sizes, at a high level we have two options:

* Like you said, pass in the offsets/sizes as arguments to the kernel. (I'm not entirely sure what's going on with `x_at_index_i` above… can you clarify what `i` is?)
* Use Dahlia to somehow generate one-off, fixed-size kernels that _only_ work for a particular size. (This would require metaprogramming on the Dahlia side. Namely, we have in the past used some crazy [m4](https://en.wikipedia.org/wiki/M4_(computer_language)) macros to automatically adjust the size for a given Dahlia kernel before compiling it.)

The former is perhaps more convenient, while the latter is perhaps more efficient.

cgyurgyik, 2020-11-02:
> * Like you said, pass in the offsets/sizes as arguments to the kernel. (I'm not entirely sure what's going on with `x_at_index_i` above… can you clarify what `i` is?)

So for this I mean, how do we encapsulate incrementing both the input arguments and the output arguments within the component that represents the function? For example, in a tensor_add of `A=[1, 2, 3]`, `B=[1, 1, 1]`,
Let's call the output `C`. Then, I want `C.addr0 = 0`, same for A and B. I compute the add, then increment, compute the add, and so on. So by `x_at_index_i`, I meant that if `C.addr0` is at index `2`, this should also be the case for `A, B`. However, for the scalar case, we simply just pass in `A_out`, `B_out`. This obviously doesn't work here. 

So maybe the fix here is to also pass in `A_addr0`, `B_addr0` as additional inputs, so we can increment those as well.

sampsyo, 2020-11-02:
> So maybe the fix here is to also pass in `A_addr0`, `B_addr0` as additional inputs, so we can increment those as well.

Yeah! Indeed, I think that is exactly the thing to do. If we do that, we will be exposing the _entire interface_ to the underlying memory to the component that needs to read/write that memory.

cgyurgyik, 2020-11-02:
Ok so from my understanding, the next steps that need to occur are in the Dahlia code. We need to:
1. Decide which approach to take.
2. Decide how to say that a memory `X` is an input or an output.
3. Make the necessary code changes in Dahlia.

Keeping in mind the overarching goal: Black box a Dahlia function, so that it produces a component with corresponding inputs and outputs. We can then lower it to FuTIL, create a group `run_foo` which hooks up the input wires/output wires in our Relay2FuTIL to the component, and runs it in a group. I may be missing some smaller things here, but I think that's the general idea.

cgyurgyik, 2020-11-03:
To further illustrate my point, here's an example of what the expected outcome might look like. This takes `examples/dahlia/vectorized-add.fuse` and separates the add in its own component (note that these manipulations were done by hand; our goal is to automate this).

```
import "primitives/std.lib";
component vector_add(A0_out: 32, B0_out: 32, in_done: 1) ->
                    (A0_addr0: 4, B0_addr0: 4, in_write_data: 32, in_write_en: 1, in_addr0: 4) {
  cells {
    A_read0_0 = prim std_reg(32);
    B_read0_0 = prim std_reg(32);
    add0 = prim std_add(32);
    add1 = prim std_add(4);
    const0 = prim std_const(4,0);
    const1 = prim std_const(4,7);
    const2 = prim std_const(4,1);
    i0 = prim std_reg(4);
    le0 = prim std_le(4);
  }
  wires {
    group cond0<"static"=0> {
      cond0[done] = 1'd1;
      le0.left = i0.out;
      le0.right = const1.out;
    }
    group let0<"static"=1> {
      i0.in = const0.out;
      i0.write_en = 1'd1;
      let0[done] = i0.done;
    }
    group upd0<"static"=1> {
      A_read0_0.write_en = 1'd1;
      A0_addr0 = i0.out;
      A_read0_0.in = 1'd1 ? A0_out;
      upd0[done] = A_read0_0.done ? 1'd1;
    }
    group upd1<"static"=1> {
      B_read0_0.write_en = 1'd1;
      B0_addr0 = i0.out;
      B_read0_0.in = 1'd1 ? B0_out;
      upd1[done] = B_read0_0.done ? 1'd1;
    }
    group upd2<"static"=1> {
      in_addr0 = i0.out;
      in_write_en = 1'd1;
      add0.left = A_read0_0.out;
      add0.right = B_read0_0.out;
      in_write_data = 1'd1 ? add0.out;
      upd2[done] = in_done ? 1'd1;
    }
    group upd3<"static"=1> {
      i0.write_en = 1'd1;
      add1.left = i0.out;
      add1.right = const2.out;
      i0.in = 1'd1 ? add1.out;
      upd3[done] = i0.done ? 1'd1;
    }
  }
  control {
    seq {
      let0;
      while le0.out with cond0 {
        seq {
          par {
            upd0;
            upd1;
          }
          upd2;
          upd3;
        }
      }
    }
  }
}

component main() -> () {
  cells {
    A0 = prim std_mem_d1_ext(32,8,4);
    B0 = prim std_mem_d1_ext(32,8,4);
    Sum0 = prim std_mem_d1_ext(32,8,4);
    Vadd0 = vector_add;
  }
  wires {
    group run_vectorized_add {
      A0.addr0 = Vadd0.A0_addr0;
      B0.addr0 = Vadd0.B0_addr0;
      Vadd0.B0_out = B0.read_data;
      Vadd0.A0_out = A0.read_data;

      Vadd0.in_done = Sum0.done;
      Sum0.addr0 = Vadd0.in_addr0;
      Sum0.write_data = Vadd0.in_write_data;
      Sum0.write_en = Vadd0.in_write_en;

      Vadd0.go = 1'd1;
      run_vectorized_add[done] = Vadd0.done ? 1'd1;
    }
  }
  control {
    seq {
      run_vectorized_add;
    }
  }
}
```

This has the expected behavior of conducting a vectorized add. So what did I change? For each input `A0, B0`, I needed the following:
1. input wire *_out
2. output wire *_addr0

For the output, I needed to add:
1. input wire *_done.
2. output wire *_write_data.
3. output wire *_write_en.
4. output wire *_addr0 (for tensors > 1-dimensional, we'll need to add subsequent addresses as well).

sgpthomas, 2020-11-03:
this is the right idea. with memories, you can automate some of this process by using `std_mem_d*_ext`. When you use this memory component and run the `externalize` pass in the FuTIL compiler. it will automatically move the interface of the sub-component into the signature of the component. If you need to do this with registers, then the way to do it is to add a `std_reg_ext` that works in the same way.

w.r.t the second point, I would recommend just using Dahlia's current system of only having inputs. These are the memory declarations that begin with `decl`. Dahlia then places no restrictions on how `decl`s are used. It's up to what ever is calling the Dahlia kernel to pass in all the memories that are required and then reading from the memories where it expects the output to be.

To illustrate this, consider a program that wants to use Dahlia to generate a vector add kernel. Logically, vector add has the type `vec<n>. vec<n> -> vec<n>`. But instead of explicitly separating input from output, we'll write the following Dahlia program:
```
decl A: float[32]; // implicitly an input
decl B: float[32]; // implicitly an input
decl C: float[32]; // the output
...
// code implementing the vector add
```
then in the program that is using Dahlia to generate kernels, using python-like code:
```
a = [1] * 32;
b = [2] * 32;
c = [0] * 32;
kernel(a, b, c); // pass in all memories
// now c should contain output
assert(c == [3] * 32)
```

Another way to think about this approach is that the kernel contains no information about the inputs and outputs of the kernel. Instead, the wrapping program generating the kernel needs to store this information separately and then use this information to figure out what the inputs and outputs are.

cgyurgyik, 2020-11-03:
Huh ok interesting. So using an external memory just basically throws all wires of the memory into the component's inputs and outputs, regardless of whether it actually is an input or output. Thanks for the explanation.

closed 2020-11-08 by cgyurgyik