cucapra/dahlia #329: index_address sizes and 3-dimensional matrices

cgyurgyik, 2020-11-09:
Has anyone run into the following before?
```
decl x: ubit<32>[1][2][2];
decl y: ubit<32>[1][2][2];
decl res: ubit<32>[1][2][2];

for (let batch: ubit<1> = 0..1) {
  for (let i: ubit<2> = 0..2) {
    for (let j: ubit<2> = 0..2) {
      for (let k: ubit<2> = 0..2) {
        let product = res[batch][i][k] * y[batch][k][j];
      } combine {
        res[batch][i][j] += product;
      }
    }
  }
}
```
^ This works fine if I make everything 2-dimensional. I’ve confirmed the semantics with data as well. However, when I add the batch dimension, I get the following error:

```
Stage 'verilog' had a non-zero exit code.
---------------- stderr ----------------
%Warning-WIDTH: /var/folders/92/gt5cgc1d6rz4r1yvkl_p9www0000gn/T/tmpj7ikzhpc:40:32: Bit extraction of array[1:0] requires 1 bit index, not 2 bits.
                                                                                  : ... In instance main.res0_0_0
   40 |   assign read_data = mem[addr0][addr1][addr2];
      |                                ^
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Warning-WIDTH: /var/folders/92/gt5cgc1d6rz4r1yvkl_p9www0000gn/T/tmpj7ikzhpc:40:39: Bit extraction of array[1:0] requires 1 bit index, not 2 bits.
                                                                                  : ... In instance main.res0_0_0
   40 |   assign read_data = mem[addr0][addr1][addr2];
      |                                       ^
%Warning-WIDTH: /var/folders/92/gt5cgc1d6rz4r1yvkl_p9www0000gn/T/tmpj7ikzhpc:43:17: Bit extraction of array[1:0] requires 1 bit index, not 2 bits.
                                                                                  : ... In instance main.res0_0_0
   43 |       mem[addr0][addr1][addr2] <= write_data;
      |                 ^
%Warning-WIDTH: /var/folders/92/gt5cgc1d6rz4r1yvkl_p9www0000gn/T/tmpj7ikzhpc:43:24: Bit extraction of array[1:0] requires 1 bit index, not 2 bits.
                                                                                  : ... In instance main.res0_0_0
   43 |       mem[addr0][addr1][addr2] <= write_data;
      |                        ^
%Error: Exiting due to 4 warning(s)
```

So if I entertain the error messages, i.e.:
```
decl x: ubit<32>[1][2][2];
decl y: ubit<32>[1][2][2];
decl res: ubit<32>[1][2][2];

for (let batch: ubit<1> = 0..1) {
  for (let i: ubit<1> = 0..2) {
    for (let j: ubit<1> = 0..2) {
      for (let k: ubit<1> = 0..2) {
        let product = res[batch][i][k] * y[batch][k][j];
      } combine {
        res[batch][i][j] += product;
      }
    }
  }
}
```

I get:
```
Stage 'dahlia' had a non-zero exit code.
---------------- stderr ----------------
[WARN] [9.27] ubit<1> is used for an array access. This might be out of bounds at runtime.
        let product = res[batch][i][k] * y[batch][k][j];
                          ^

[WARN] [9.44] ubit<1> is used for an array access. This might be out of bounds at runtime.
        let product = res[batch][i][k] * y[batch][k][j];
                                           ^

[WARN] [11.13] ubit<1> is used for an array access. This might be out of bounds at runtime.
        res[batch][i][j] += product;
            ^

[Error] [Line 0, Column 0] Cast implies zero-padding from 1 to 2 bits
<undefined position>
 This feature is not yet implemented. Please open a feature request for it.
```

I'm unsure where the underlying issue is here, or where to start looking. Any pointers in this regard are welcome.

rachitnigam, 2020-11-09:
Updated FuTIL master so you don't get the first error message. It's a linting warning that needed to be turned off.

cgyurgyik, 2020-11-09:
> Updated FuTIL master so you don't get the first error message. It's a linting warning that needed to be turned off.

Thanks Rachit!

closed 2020-11-09 by cgyurgyik