cucapra/dahlia #348: Add memory to Function Definitions for Futil backend.

cgyurgyik, 2021-02-01:
The naive way is to hook up all wires, regardless of whether we want to read or write to the memory. 

A more restrictive approach is providing connections to different ports based on some parameter `read-only, write-only, read-write, ...`.

cgyurgyik, 2021-02-01:
This may be slightly more difficult than anticipated. Since we want to pass in memory ports, e.g. for memory `x: ubit<32>[1]`,
we'll have function parameters: `x_addr0, x_read_data`. This means we need to translate the following:
```
let tmp = x[0];
```
into reading the ports defined in the function definition, instead of trying to access `x.addr0, x.read_data` as what is done in `EArrAccess`.

sampsyo, 2021-02-02:
Just linking the relevant issue at https://github.com/cucapra/futil/issues/381.

cgyurgyik, 2021-02-06:
Fixed with #349. Currently, just includes all write and read ports.

closed 2021-02-06 by cgyurgyik