cucapra/dahlia #352: Double imports for FuTIL backend

cgyurgyik, 2021-02-03:
Because of this:
https://github.com/cucapra/dahlia/blob/e2712491aefcdf5d61e6b4a78040f13602f05f42/src/main/scala/backends/futil/FutilBackend.scala#L867-L869

We're seeing:

https://github.com/cucapra/dahlia/blob/e2712491aefcdf5d61e6b4a78040f13602f05f42/file-tests/should-futil/invoke.fuse#L1-L10

which results in:

https://github.com/cucapra/dahlia/blob/e2712491aefcdf5d61e6b4a78040f13602f05f42/file-tests/should-futil/invoke.expect#L1-L3

cgyurgyik, 2021-02-03:
I think the best way to fix this is to not have a default import. This will require tediously refactoring all current Dahlia files. I'll wait for second opinions before following through.

rachitnigam, 2021-02-03:
Sqrt isn’t available in std.lib, I moved it to a different folder. Dahlia programs shouldn’t be importing that file.

cgyurgyik, 2021-02-04:
Ok so that's wrong. But in general, will it ever be the case where this results in a double import, or is it always incorrect to do so?

rachitnigam, 2021-02-04:
My guess would be no. `std.lib` should really only implement things that Dahlia always *needs* in order to work properly. For example, it exports all the primitives for fixed point numbers. OTOH, things like sqrt should really exist as only their own in a different Calyx library.

cgyurgyik, 2021-02-04:
Makes sense. Fixed in #351.

closed 2021-02-06 by cgyurgyik