cucapra/dahlia #353: Using fixed point in futil imports or function definitions doesn't work.

cgyurgyik, 2021-02-10:
### Futil imports
```
import futil("primitives/bitnum/math.futil")
{
  def std_sqrt(in: ufix<32, 16>): ufix<32, 16>;
}

let x: ufix<32, 16> = 4.0;

let t: ufix<32, 16> = std_sqrt(x);
```
->
```
ERROR: Stage 'dahlia' had a non-zero exit code.
---------------- stderr ----------------
[Error] [Line 8, Column 23] `std_sqrt` should be assigned to its own `let` statement, e.g. `let _temp = std_sqrt(...);`
let t: ufix<32, 16> = std_sqrt(x);
                      ^
 This feature is not yet implemented. Please open a feature request for it.
```

### Function definitions
```
def foo(in: ufix<32, 16>): ufix<32, 16> = {
  let temp: ufix<32, 16> = in;
  return temp;
}

let x: ufix<32, 16> = 4.0;

let t: ufix<32, 16> = foo(x);
```
->
```
ERROR: Stage 'dahlia' had a non-zero exit code.
---------------- stderr ----------------
[Error] [Line 8, Column 23] `foo` should be assigned to its own `let` statement, e.g. `let _temp = foo(...);`
let t: ufix<32, 16> = foo(x);
                      ^
 This feature is not yet implemented. Please open a feature request for it.
```

rachitnigam, 2021-02-10:
Use `-l debug` to get the stack trace and see which line in the compiler throws this error.

cgyurgyik, 2021-02-10:
This is with ```fuse a.fuse -b=futil -l=debug```:
```
[Error] [Line 8, Column 22] `foo` should be assigned to its own `let` statement, e.g. `let _temp = foo(...);`
let t: fix<32, 16> = foo(x);
                     ^
 This feature is not yet implemented. Please open a feature request for it.
```

rachitnigam, 2021-02-10:
Not sure which part of the compiler is throwing it. Probably the code for EApp in futilbackend. That flag should’ve dumped the stack trace. See the CLI flags of the compiler to make sure it’s not log-level or something

PR #354 (2021-02-10): Fix `CLet` match ordering.
Originally, an earlier case `CLet(id, Some(TFixed(t, i, un)), Some(e)) => {` was being matched before the more specific criteria where `e == EApp`.

Fixes #353.
closed 2021-02-10 by rachitnigam