cucapra/dahlia #427: Calyx back-end issues

Mark1626, 2024-03-21:
I noted these two issue when I was trying to add float support in Dahlia (Calyx backend)

## 1.  Hoist memory pass issue

The following does not pass with `--lower` flag.

```
decl a: float[128];
decl b: bit<10>[128];
decl c: float[128];

for (let i:bit<8>=0..128) {
    c[i] := a[b[i]];
}
```

**Error**

```
[Type error] [Line 0, Column 0] `b_read0' is not bound in scope.
<undefined position>

```

Root cause seems to be the Hoist pass, `b_read0` has not been created

```
===============Original===============
decl a: float[128];
decl b: bit<10>[128];
decl c: float[128];
for (let i: bit<8> = 0 .. 128) {
  c[i] := a[b[i]];
}
======================================
===============Hoist memory reads===============
decl a: float[128];
decl b: bit<10>[128];
decl c: float[128];
for (let i: bit<8> = 0 .. 128) {
  let a_read0 = a[b_read0];
  c[i] := a_read0;
}

```

---

## 2. Incorrect usage of signed Calyx primitives

The following generates Calyx IR

```
decl a: bit<10>[128];
decl b: bit<10>[128];
decl c: bit<10>[128];

for (let i:bit<8>=0..128) {
    c[i] := a[i] & b[i];
}

```

The Calyx IR contains a `std_sand` primitive which seems to be not present in the current Calyx primitives 

```
import "primitives/core.futil";
import "primitives/memories/seq.futil";
import "primitives/binary_operators.futil";
import "P32.futil";
component main() -> () {
  cells {
    @external(1) a = seq_mem_d1(10,128,8);
    a_read0_0 = std_reg(10);
    add0 = std_sadd(8);
    and0 = std_sand(10);
   ...
   ...
   ...
```

rachitnigam, 2024-03-21:
Thanks! (1) seems like a bug in lowering while (2) is probably in the backend. For (2), we basically attempt to generate the string name of the operator based on whether or not the operands are signed or not (adding the `s` if they are). For logical operators like `&`, I think we should probably not generate the `s`?

Mark1626, 2024-03-21:
2. Seems to be because of this function. I feel the Calyx backend code needs some restructuring, we seem to lose information about the operator

https://github.com/cucapra/dahlia/blob/9ec9a58fa5fefc4e3fc5e19a38d90c3f780ec4b8/src/main/scala/backends/calyx/Ast.scala#L366-L370

I'll try to address both these issues in a PR when I get some time